espressif/esp-nn #1: Is there perfomance gain for plain ESP32?

AIWintermuteAI, 2022-02-21:
Very exciting development!
I see benchmark numbers comparing ANSI C vs Assembly optimized for ESP32-S3. Have you tried comparison benchmarks for ESP-NN ANSI C vs vanilla tflite-micro ops? Is there any performance gain in that case for ESP32 (plain, not S3 version)?

vikramdattu, 2022-02-21:
Hi @AIWintermuteAI 

> Very exciting development!
I see benchmark numbers comparing ANSI C vs Assembly optimized for ESP32-S3. Have you tried comparison benchmarks for ESP-NN ANSI C vs vanilla tflite-micro ops? Is there any performance gain in that case for ESP32 (plain, not S3 version)?

Thanks for the interest!
The gain is about 2x when ANSI C versions of ESP-NN are compared with tflite-ops with no changes.
[This](https://github.com/espressif/tflite-micro-esp-examples/blob/master/components/tflite-lib/CMakeLists.txt#L61) flag when commented out, uses vanilla tflite-micro kernels. When enabled, it uses ANSI C versions OR assembly optimized ESP32-S3.

AIWintermuteAI, 2022-02-21:
Thanks for lightning speed reply!
Excited to use ESP-NN then. I'll close the issue.

closed 2022-02-21 by AIWintermuteAI