espressif/esp-nn #2: Releases

AIWintermuteAI, 2022-04-04:
Hello!
I've tested optimized kernels for plain ESP32 and got very nice performance boost! Good job!
A little organizational question - do you plan to have releases published? For my project, all the components will be automatically obtained with ```git clone``` and in order to ensure stable builds either release or branch number is preferred. I could use a commit number too, but that would be less convenient. I'm sure other people using ESP-NN also would appreciate having versioned releases.

vikramdattu, 2022-04-05:
Hi @AIWintermuteAI release sure is under consideration and will be tagged soon.

nliaudat, 2022-12-12:
any up ?

vikramdattu, 2022-12-13:
Hi @nliaudat you can use following tag: [v1.0-beta](https://github.com/espressif/esp-nn/releases/tag/v1.0-beta)

kedars, 2022-12-13:
https://github.com/espressif/esp-nn/releases/tag/v1.0-beta

closed 2022-12-13 by kedars

nliaudat, 2022-12-13:
many thanks !
Regards