espressif/esp-nn #3: Why ESP32-S3 in ESP-IDF v4.4.2 building error?

WangShuoran, 2022-10-14:
I want to use this repo to build my fully_connected application. But when I use my ESP32-S3 and ESP-IDF v4.4.2 to build this application will report error.

[CMakeOutput.log](https://github.com/espressif/esp-nn/files/9783967/CMakeOutput.log)

vikramdattu, 2022-10-14:
@WangShuoran 
Few questions:
1. Does the[ test application](https://github.com/espressif/esp-nn/tree/master/test_app) build well for you?  
2. What method do you use to include `esp-nn` in your project? Looks like your application is trying to search for components insider `esp-nn/` directory whereas, `esp-nn` itself is a component

Additionally, can you please also share console logs?

WangShuoran, 2022-10-14:
Sorry, I forget it.
I am use repo/test_app(as your 1. say), I think this is an example, when I inter the test_app folder , select ESP32-S3, and run idf.py menuconfig. the error console is:
```console
{}[SDK Configuration Editor]
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.36.1.windows.1") 

[SDK Configuration Editor]
-- ccache will be used for faster recompilation

[SDK Configuration Editor]
-- The C compiler identification is GNU 8.4.0

[SDK Configuration Editor]
-- The CXX compiler identification is GNU 8.4.0

[SDK Configuration Editor]
-- The ASM compiler identification is GNU

[SDK Configuration Editor]
-- Found assembler: C:/Espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc.exe

[SDK Configuration Editor]
-- Detecting C compiler ABI info

[SDK Configuration Editor]
-- Detecting C compiler ABI info - done

[SDK Configuration Editor]
-- Check for working C compiler: C:/Espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc.exe - skipped

[SDK Configuration Editor]
-- Detecting C compile features

[SDK Configuration Editor]
-- Detecting C compile features - done

[SDK Configuration Editor]
-- Detecting CXX compiler ABI info

[SDK Configuration Editor]
-- Detecting CXX compiler ABI info - done

[SDK Configuration Editor]
-- Check for working CXX compiler: C:/Espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-g++.exe - skipped

[SDK Configuration Editor]
-- Detecting CXX compile features

[SDK Configuration Editor]
-- Detecting CXX compile features - done

[SDK Configuration Editor]
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32s3

[SDK Configuration Editor]
---------------------------ERROR--------------------------

CMake Error at C:/Espressif/frameworks/esp-idf-v4.4.2/tools/cmake/build.cmake:201 (message):
  Failed to resolve component 'esp-nn'.
Call Stack (most recent call first):
  C:/Espressif/frameworks/esp-idf-v4.4.2/tools/cmake/build.cmake:232 (__build_resolve_and_add_req)
  C:/Espressif/frameworks/esp-idf-v4.4.2/tools/cmake/build.cmake:233 (__build_expand_requirements)
  C:/Espressif/frameworks/esp-idf-v4.4.2/tools/cmake/build.cmake:509 (__build_expand_requirements)
  C:/Espressif/frameworks/esp-idf-v4.4.2/tools/cmake/project.cmake:384 (idf_build_process)
  CMakeLists.txt:9 (project)

-----------------------END OF ERROR-----------------------
[SDK Configuration Editor]
-- Configuring incomplete, errors occurred!
See also "D:/test_study_v2/esp-nn-master/test_app/build/CMakeFiles/CMakeOutput.log".
Executing action: confserver
Running cmake in directory d:\test_study_v2\esp-nn-master\test_app\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DCCACHE_ENABLE=1 d:\test_study_v2\esp-nn-master\test_app"...

[SDK Configuration Editor]
---------------------------ERROR--------------------------

cmake failed with exit code 1

-----------------------END OF ERROR-----------------------
[SDK Configuration Editor]
---------------------------ERROR--------------------------

SDK Configuration editor confserver process exited with code: 2
-----------------------END OF ERROR-----------------------

```

WangShuoran, 2022-10-14:
And I don't find the README.md let me config global environment. And the ASM I don't know how to let CMakeLists.txt import to Project, so I don't know how to config an instead CMakeLists.txt.

vikramdattu, 2022-10-14:
Thanks for the updates. 
I see that you have renamed the project folder to `esp-nn-master`. You need to keep the folder name to `esp-nn`.
This is because, the the lib gets built with the folder name!

In brief: Rename the outer folder back to `esp-nn` and the build shall go through.

For configuration you may refer to this: https://github.com/espressif/esp-nn#configuration 

Using a CMake Project is quite standard. You will need to specify requirements and include the project directory. You may refer to tesp_app for this.

vikramdattu, 2022-10-14:
@WangShuoran by the way, if you are looking for tflite-micro integration, you can already find it here: https://github.com/espressif/tflite-micro-esp-examples

WangShuoran, 2022-10-15:
I test the example, and build is ok, but when I efuse ESP32-S3 it will output:

```output
I (43) boot: ESP-IDF v4.4.2-dirty 2nd stage bootloader
I (43) boot: compile time 09:33:21
I (43) boot: chip revision: 0
I (56) boot.esp32s3: Boot SPI Speed : 80MHz
I (56) boot.esp32s3: SPI Mode       : DIO
I (57) boot.esp32s3: SPI Flash Size : 8MB
I (59) boot: Enabling RNG early entropy source...
I (65) boot: Partition Table:
I (68) boot: ## Label            Usage          Type ST Offset   Length
I (76) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (83) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (91) boot:  2 factory          factory app      00 00 00010000 00100000
I (98) boot: End of partition table
I (102) esp_image: segment 0: paddr=00010020 vaddr=3c030020 size=08248h ( 33352) map
I (117) esp_image: segment 1: paddr=00018270 vaddr=3fc91db0 size=02b40h ( 11072) load
I (122) esp_image: segment 2: paddr=0001adb8 vaddr=40374000 size=05260h ( 21088) load
I (132) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=2075ch (132956) map
I (160) esp_image: segment 4: paddr=00040784 vaddr=40379260 size=08b50h ( 35664) load
I (168) esp_image: segment 5: paddr=000492dc vaddr=50000000 size=00010h (    16) load
I (174) boot: Loaded app from partition at offset 0x10000
I (174) boot: Disabling RNG early entropy source...
I (189) opi psram: vendor id : 0x0d (AP)
I (189) opi psram: dev id    : 0x02 (generation 3)
I (189) opi psram: density   : 0x03 (64 Mbit)
I (193) opi psram: good-die  : 0x01 (Pass)
I (197) opi psram: Latency   : 0x01 (Fixed)
I (202) opi psram: VCC       : 0x01 (3V)
I (207) opi psram: SRF       : 0x01 (Fast Refresh)
I (212) opi psram: BurstType : 0x01 (Hybrid Wrap)
I (218) opi psram: BurstLen  : 0x01 (32 Byte)
I (223) opi psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (229) opi psram: DriveStrength: 0x00 (1/1)
W (234) PSRAM: DO NOT USE FOR MASS PRODUCTION! Timing parameters will be updated in future IDF version.
I (244) spiram: Found 64MBit SPI RAM device
I (249) spiram: SPI RAM mode: sram 80m
I (253) spiram: PSRAM initialized, cache is in normal (1-core) mode.
I (260) cpu_start: Pro cpu up.
I (264) cpu_start: Starting app cpu, entry point is 0x40375230
0x40375230: call_start_cpu1 at C:/Espressif/frameworks/esp-idf-v4.4.2/components/esp_system/port/cpu_start.c:160

I (0) cpu_start: App cpu up.
I (559) spiram: SPI SRAM memory test OK
I (567) cpu_start: Pro cpu start user code
I (567) cpu_start: cpu freq: 240000000
I (567) cpu_start: Application information:
I (570) cpu_start: Project name:     test_app
I (575) cpu_start: App version:      1
I (579) cpu_start: Compile time:     Oct 15 2022 09:33:12
I (585) cpu_start: ELF file SHA256:  5fbba61455b970c4...
I (591) cpu_start: ESP-IDF:          v4.4.2-dirty
I (597) heap_init: Initializing. RAM available for dynamic allocation:
I (604) heap_init: At 3FC95308 len 0004ACF8 (299 KiB): D/IRAM
I (610) heap_init: At 3FCE0000 len 0000EE34 (59 KiB): STACK/DRAM
I (617) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
I (623) spiram: Adding pool of 8192K of external SPI memory to heap allocator
I (631) spi_flash: detected chip: winbond
I (636) spi_flash: flash io: dio
I (640) sleep: Configure to isolate all GPIO pins in sleep state
I (647) sleep: Enable automatic switching of GPIO sleep configuration
I (654) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (674) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (674) test_app: Running s8 tests...
esp_nn_add_elementwise_s8_test[0] passed
esp_nn_add_elementwise_s8_test[1] passed
esp_nn_add_elementwise_s8_test[2] passed
esp_nn_add_elementwise_s8_test[3] passed
[... 55 lines cut ...]

0x4037c171: vPortTaskWrapper at C:/Espressif/frameworks/esp-idf-v4.4.2/components/freertos/port/xtensa/port.c:131

ELF file SHA256: 5fbba61455b970c4

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0xa (SPI_FAST_FLASH_BOOT)
Saved PC:0x40375848
0x40375848: esp_restart_noos at C:/Espressif/frameworks/esp-idf-v4.4.2/components/esp_system/port/soc/esp32s3/system_internal.c:143 (discriminator 1)

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x1654
load:0x403c9700,len:0xbb8
load:0x403cc700,len:0x2f88
SHA-256 comparison failed:
Calculated: 26be4cebc6558f2beaabc506afee9df1c5d97fdfa86af8bd2b9bf7868ea91386
Expected: 2bbb1f86d0b09daca94e77b08476226fd42c131cac433e50ef9a3bafa8178d8c
Attempting to boot anyway...
entry 0x403c9954
```

I think the "s8 tests done!" is not output explain the code is error.

vikramdattu, 2022-10-15:
Hi @WangShuoran the allocation is failing for the large block!

Please enable SPIRAM for the test to run.
You can do this via menuconfig: `idf.py menuconfig, component configs, ESP32-S3 specific, SPIRAM`.

WangShuoran, 2022-10-15:
So can I think the `esp_nn_softmax_s8_test()` need the SPIRAM? But another API do not need SPIRAM.

And when I open the menuconfig “CONFIG_SPIRAM_IGNORE_NOTFOUND”, the ESP32-S3 will error with my before answer. I do not know way.

Thank you for a lot answer. I am running this repo good. I think the repo is very useful.

vikramdattu, 2022-10-15:
The test specifically fails to allocate as the chunk it requires is quite large!
About SPIRAM,  you seem to have already enabled it:
```
I (244) spiram: Found 64MBit SPI RAM device
I (249) spiram: SPI RAM mode: sram 80m
I (253) spiram: PSRAM initialized, cache is in normal (1-core) mode.
I (260) cpu_start: Pro cpu up.
I (264) cpu_start: Starting app cpu, entry point is 0x40375230
0x40375230: call_start_cpu1 at C:/Espressif/frameworks/esp-idf-v4.4.2/components/esp_system/port/cpu_start.c:160
```
I didn't notice the logs before.

Can you please also make sure that it is allocatable? 
The code shouldgo through this path: https://github.com/espressif/esp-nn/blob/master/tests/src/convolution_test.c#L25

WangShuoran, 2022-10-17:
As you know, I am exactly enable the SPIRAM, and if I enable the CONFIG_SPIRAM_IGNORE_NOTFOUND, this example will trigger error, error in the answer above.

If I disable the CONFIG_SPIRAM_IGNORE_NOTFOUND, the error will disappear.

So, I don't know why, and I can not make sure why enable the CONFIG_SPIRAM_IGNORE_NOTFOUND that it is allocatable.

WangShuoran, 2022-10-17:
I think CONFIG_SPIRAM_IGNORE_NOTFOUND is close in release because of consistency assurance, so I think this option is only a small probability event, so I do not want to reproduce this error.

But I have another question. You know I am use this repo's test_app to learn this repo, but if I want to integrate this repo into my project, components is in repo (dir: repo/components).
(One option is) I need copy esp-nn into components, and in my main.c to include esp-nn include file and call esp-nn API.
Does this is OK?

And you say:

> Thanks for the updates. I see that you have renamed the project folder to `esp-nn-master`. You need to keep the folder name to `esp-nn`. This is because, the the lib gets built with the folder name!

Does I cannot in my project/components rename this repo?

vikramdattu, 2022-10-17:
Hi @WangShuoran you can keep the esp-nn repo at any location. You would also need to specify it in the project CMakeList.txt as extra component dir.

You can also rename the repo, however, while specifying dependencies, you would specify new name. (esp-nn-master in place of esp-nn).

For more info on IDF build system, please follow: https://espressif-docs.readthedocs-hosted.com/projects/esp-idf/en/latest/api-guides/build-system.html

WangShuoran, 2022-10-17:
Hi @vikramdattu thank you give me those info, I think I am totally understand what you mean.

To make sure I fully understand what you mean, I will restate your words:The esp-nn as a component can be rename with any another name(almost).

You say: 
> I see that you have renamed the project folder to esp-nn-master. You need to keep the folder name to esp-nn.
This is because, the the lib gets built with the folder name!
> In brief: Rename the outer folder back to esp-nn and the build shall go through.

But why the repo/test_app should name this repo with `esp-nn` instead of `esp-nn-master`? I am in test_app CMakeLists.txt want to find out.

This is project CMakeLists.txt, I think in project CMakeLists.txt don't appoint `esp-nn` instead of another.
```cmake
# The following lines of boilerplate have to be in your project's
# CMakeLists in this exact order for cmake to work correctly
cmake_minimum_required(VERSION 3.5)

set(EXTRA_COMPONENT_DIRS "../" "../tests/")
set(IDF_EXCLUDE_COMPONENTS test test_app)

include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(test_app)

```

and repo/main/CMakeLists.txt is:
```cmake

set(COMPONENT_SRCS "main.c")
set(COMPONENT_ADD_INCLUDEDIRS "")

set(COMPONENT_PRIV_REQUIRES tests)

register_component()

```
I also think this don't  appoint `esp-nn` instead of another.

So, I don't know why in test_app should rename `esp-nn` instead of `esp-nn-master`. Exactly, I do not know "the lib gets built with the folder name!" in test_app where?

WangShuoran, 2022-10-18:
I think I am completely figured out how to use this repo. Thank you for a lot!

closed 2022-10-18 by WangShuoran