espressif/esp-nn #9: Can I use `esp-nn/src/convolution /esp_nn_conv_s8_mult8_1x1_esp32s3.S` in my own project?

junnan-xjtu, 2023-12-14:
I want to deploy some open source algorithms on esp32s3. The operators used in the current deployment are all designed by myself and are very simple. For example, convolution is nested through for loops, combined with some img2col processing, and the operating efficiency is very low.

I saw that there are some assembly source codes in this warehouse that use esp32s3 extended instructions for computing acceleration. Can I directly migrate these assembly operators to my project? If so, what adaptation work should I do? If not, can you tell me how to use the extended instructions of esp32s3? I don't have much experience in using assembly language.

Thank you so much!

vikramdattu, 2023-12-14:
Hello @junnan-xjtu you may refer to my assembly and modify the same to cater your needs.
You may take a look at following trm which will help you understand and write the assembly: https://www.espressif.com/sites/default/files/documentation/esp32-s3_technical_reference_manual_en.pdf

junnan-xjtu, 2023-12-14:
Thank you for your prompt reply!
I understand what you mean, and that's exactly what I plan to do. But I saw you mentioned that `esp-nn` supports tflite APIs only, which is shown in `https://github.com/espressif/esp-nn/issues/4`. In my project, the model weights are all exported into one-dimensional arrays instead of tflite-micro format. Will this have any impact?

vikramdattu, 2026-11-09:
Closing, as this was answered above. The ESP-NN functions take plain arrays and dimension structs, so they do not depend on the TFLite model format.

closed 2026-11-09 by vikramdattu