espressif/esp-nn #11: Support for ESP32-P4?

AIWintermuteAI, 2024-04-15:
Hello, @vikramdattu !
Do you have any information about how NN accelerated inference is going to be handled in the new P4 chip? Will it be included in ESP NN as well?

vikramdattu, 2024-04-15:
@AIWintermuteAI yes, the optimizations will be included in esp-nn for esp32-p4.
Once added, these will get included transparently added to tflite-micro.
Unfortunately, I do not have any ETA at the moment though.

nicklasb, 2024-08-25:
Oh, I just saw this here, great stuff.

nicklasb, 2024-09-15:
@vikramdattu 
Hi, testing with an ESP32-P4 dev board (320 Mhz) and realized that there probably are no P4-specific optimizations enabled?

Because I have an inference (YOLOV5) that takes ESP32-S3 40-50 seconds.
However, the ESP32-P4 using defaults (128 K L2 cache) take 100 seconds. With 512 L2 we are down to 63 seconds.

I don't remember exacly the performance of the S3 unoptimized, but It feels like 100 is roughly how much faster I would expect the P4 would be unoptimized?

vikramdattu, 2024-09-16:
@nicklasb we have just pushed the first-cut support for ESP32-P4.
 - Enabled generic optimisations from esp-nn for esp32p4 chip
 - Convolution function is optimised using inline assembly

With these optimisations you should see 2x-7x performance boost depending on the model architecture. You should see yet better version for convolution and also optimisations for other kernels as well in follow-up releases.

nicklasb, 2024-09-16:
@vikramdattu 
So now it is down to 35 seconds, so some improvement, certainly, however no where near 2x-7x, so as of yet, it does less per cycle than the S3, so not super impressed yet. ;-)

Are there some other settings I should consider, or is just the code very immature?

For context, these are the operations. It is kind of typical for a simple YOLOv5 ([later YOLO versions kees the making the models hybrid ](https://github.com/ultralytics/ultralytics/issues/16090) model.

```
    resolver.AddConv2D();
    resolver.AddDepthwiseConv2D();
    resolver.AddAdd();
    resolver.AddConcatenation();
    resolver.AddResizeNearestNeighbor();
    resolver.AddRelu();
    resolver.AddPad();
    resolver.AddLogistic();
    resolver.AddMul();
    resolver.AddMaxPool2D();
    resolver.AddReshape();
    resolver.AddStridedSlice();
    resolver.AddQuantize();
```

I am using TFLite Micro, the model has one class, is about 2 mb and is quantized to int8.
I am not sure if I would make any large gains by using ESP-DL on this one.

vikramdattu, 2024-09-17:
Hi @nicklasb you're right.
This is just a start. The optimizations are at an early stage, and that too only conv function optimized right now.
This will definitely be better and surpass S3 in a week or two.

nicklasb, 2024-09-17:
> Hi @nicklasb you're right. This is just a start. The optimizations are at an early stage, and that too only conv function optimized right now. This will definitely be better and surpass S3 in a week or two.
Great, if you report on what operations are being optimized I can helt out with up performance numbers here. 
Because from what I've gathered from the marketing material, I am expecting it to end up being about 40 times faster than the S3. :-)

vikramdattu, 2024-09-17:
> Great, if you report on what operations are being optimized I can help out with up performance numbers and profiling here. Because from what I've gathered from the marketing material, I am expecting it to end up being about 40 times faster than the S3. :-)

I will keep you posted on the progress. 
BTW, can you share with me the link for the S3 vs P4 documentation you've referred to? I have missed this particular point. Would help me understand the premise for the claim.

nicklasb, 2024-09-17:
> BTW, can you share with me the link for the S3 vs P4 documentation you've referred to? I have missed this particular point. Would help me understand the premise for the claim.

I was being a bit ironic, but if you market something with "AI extensions", one expects that the extension will provide something that won't be that much slower than much older chips, like existing edge TPUs. I think few consider "AI" be much less than being able to run deep learning models at a reasonable frame rate.

nicklasb, 2024-09-18:
Hi!

Ok, so that was faster, 29 seconds, about 6 seconds faster.
However, the boxes change a bit, the confidence seem to lower slightly with the patch.
It am filtering confidence at 0.7 and have an IOU threshold at 0.1:

```
# P4 v1 - the one in master
boxes = [
    [295.74, 189.43, 304.70, 207.35, 0.78],
    [295.74, 186.44, 304.70, 210.34, 0.71],
    [295.74, 189.43, 304.70, 207.35, 0.80],
    [221.81, 198.39, 226.29, 210.34, 0.71],
    [322.63, 180.47, 474.98, 204.36, 0.80],
    [318.15, 180.47, 479.46, 204.36, 0.77],
]

# P4 Patch 2024-09-18
boxes = [
    [295.74, 189.43, 304.70, 207.35, 0.76],
    [295.74, 189.43, 304.70, 207.35, 0.78],
    [322.63, 180.47, 474.98, 204.36, 0.78],
    [318.15, 180.47, 479.46, 204.36, 0.77],
]

```

Don't ask me to debug the code, haven't done any major assembler work since my C64 days, about 40 years ago. :-)

nicklasb, 2024-09-18:
WRT RELU, I'll have to retrain it then, I'll be back with that next week, have a lot of stuff this weekend.

nicklasb, 2024-09-22:
Annoyingly, something has broken in my Yolov5n export and int8 quantization which breaks the model in some way I have yet to be able to fix, I can't successfully export _any_ yolov5 model currently, ReLU() or not.  (and the newer versions doesn't export a pure int8 model leading to TFLite complaining on "not supporting a hybrid model"). 
I will probably have to use an other architecture. 
What is it that you are using, sounded like a similar size model?

vikramdattu, 2024-09-23:
Hi @nicklasb I used an existing model at GH user already exported from [here](https://github.com/muhammedakyuzlu/yolov5-tflite-cpp) which fits our requirement.

I  used the guide [here](https://docs.ultralytics.com/modes/export/#usage-examples) to achieve the 2.7MB model with official yolov5-n model.

```python
from ultralytics import YOLO

# Load a model
model = YOLO("yolov5n.pt")  # load an official model

# Export the model
model.export(format="tflite", int8=True)
```

One thing I noticed, if you can afford to use input image size smaller (640-->320), this should speed up model

nicklasb, 2024-09-23:
> Hi @nicklasb I used an existing model at GH user already exported from [here](https://github.com/muhammedakyuzlu/yolov5-tflite-cpp) which fits our requirement.
> 

Thanks. It seems exactly like what I have. I think that I have probably managed to screw up my Python installation or something. 
> One thing I noticed, if you can afford to use input image size smaller (640-->320), this should speed up model

Yeah, I know, but I need every pixel I can get in there, though. However, I can go grayscale, which also should help a bit, event though yolo is typically RGB.

nicklasb, 2024-09-24:
Ok, so now I have retrained with act: nn.ReLU() in the yaml settings. 
I *hope* that it actually used ReLU, I couldn't see any difference in the output but the resulting boxes are quite different:

```
boxes = [
    [292.64, 184.16, 304.75, 205.69, 0.82],
    [292.64, 184.16, 304.75, 205.69, 0.84],
    [294.66, 184.16, 302.73, 205.69, 0.88],
    [292.64, 186.86, 304.75, 203.00, 0.72],
    [294.66, 184.16, 302.73, 205.69, 0.72],
    [224.02, 194.93, 228.06, 205.69, 0.72],
    [254.29, 200.31, 262.36, 211.07, 0.75],
]
```
It doesn't "lose" boxes that a new training run with non-ReLU does:
```
boxes = [
    [293.88, 186.42, 304.83, 205.89, 0.84],
    [293.88, 186.42, 304.83, 205.89, 0.85],
    [293.88, 186.42, 304.83, 205.89, 0.87],
]
```
They both have the same training. 
And they took almost exactly the same time, so I am not sure if using ReLU made any significant difference.
(either that or I don't use it in some way)