espressif/esp-nn #12: Ocr model that trained by yolov5

WrinkLeeTR, 2024-04-18:
Hi Everyone,

I have same models that trained with yolov5 but some parameters was changed. My question is can I include OCR model that was trained with yolov5?

WrinkLeeTR, 2024-04-18:
@vikramdattu thanks for your reply. I used tflite micro and I can include my model. But I got error with esp32s3 when the devices start to run. I always got memory error. The tflite file has 1mb size. How can I solve that problem

vikramdattu, 2024-04-18:
In the `main_functions.cc`° file, you can increase TensorArena size and allocate the buffer on SPIRAM.
s3-eye has 4MB of SPIRAM.

nicklasb, 2024-08-25:
@WrinkLeeTR  
 I have it working (sloowly) with a 2 MB model, note that the for you, the tensor arena will probably take more like 2,5 MB in SPIRAM.  i needed to allocate 5 MB and therefor needed to use the S3 as the others cannot allocate more than < 4Mb in one block.

vikramdattu, 2026-11-09:
Closing, as this was answered above: a larger model needs a larger tensor arena, allocated in PSRAM.

closed 2026-11-09 by vikramdattu