espressif/esp-nn #14: Running tests multiple times crashes

maxgerhardt, 2024-06-29:
The test functions don't survive being called twice in a row (see test_app):

```
######## Running esp_nn_fully_connected_s8_test ##########
[  0] passed [row_len 271, out_ch 3]    cycles: c    30023, opt    23678
[  1] passed [row_len 271, out_ch 3]    cycles: c    23684, opt    22420
[  2] passed [row_len 271, out_ch 3]    cycles: c    23680, opt    22416
[  3] passed [row_len 271, out_ch 3]    cycles: c    23687, opt    22422
[  4] passed [row_len 1, out_ch 16]     cycles: c     3884, opt     2620
[  5] passed [row_len 16, out_ch 8]     cycles: c     5854, opt     4590
[  6] passed [row_len 8, out_ch 8]      cycles: c     4120, opt     2856
[  7] passed [row_len 8, out_ch 15]     cycles: c     6558, opt     5295
[  8] passed [row_len 8, out_ch 1]      cycles: c     1683, opt      419
[  9] passed [row_len 7, out_ch 8]      cycles: c     3902, opt     2638
[ 10] passed [row_len 7, out_ch 8]      cycles: c     3909, opt     2644
[ 11] passed [row_len 5, out_ch 8]      cycles: c     3468, opt     2204
[ 12] passed [row_len 3, out_ch 8]      cycles: c     3044, opt     1780
[ 13] passed [row_len 7, out_ch 8]      cycles: c     3906, opt     2642
[ 14] passed [row_len 2, out_ch 8]      cycles: c     2827, opt     1564

[...]
######## Running esp_nn_fully_connected_s8_test ##########
[  0] passed [row_len 2, out_ch 8]      cycles: c     7256, opt     1550
[  1] passed [row_len 2, out_ch 8]      cycles: c     2831, opt     1568
[  2] passed [row_len 2, out_ch 8]      cycles: c     2815, opt     1552
[  3] passed [row_len 2, out_ch 8]      cycles: c     2834, opt     1570
[  4] passed [row_len 1, out_ch 16]     cycles: c     3891, opt     2626
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400d7076  PS      : 0x00060e30  A0      : 0x800d2b44  A1      : 0x3ffbe3e0
A2      : 0x3ffbe460  A3      : 0x00000000  A4      : 0xffffffe7  A5      : 0x3ffbe450
A6      : 0x00000000  A7      : 0xf4e2128d  A8      : 0x0000007f  A9      : 0x00000010
A10     : 0x00000000  A11     : 0xf1aa71b7  A12     : 0xf4e2128d  A13     : 0x00000000
A14     : 0x824880e2  A15     : 0xc24880e1  SAR     : 0x00000006  EXCCAUSE: 0x0000001d  
EXCVADDR: 0xf4e2128d  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff

Backtrace: 0x400d7073:0x3ffbe3e0 0x400d2b41:0x3ffbe410 0x400d133e:0x3ffbe4a0 0x400d4fb9:0x760cc100 |<-CORRUPTED
```
Plus: I am already running at `CONFIG_ARDUINO_LOOP_STACK_SIZE=32768` to prevent it being some weird out of stack memory issue.

vikramdattu, 2024-07-01:
[0001-Fix-multiple-issues-with-tests.patch](https://github.com/user-attachments/files/16048945/0001-Fix-multiple-issues-with-tests.patch)

Hello @maxgerhardt please find a patch attached. This should fix both the issues #13 and #14
Do apply the patch and let me know if this fixes it for you.

vikramdattu, 2025-09-01:
This was fixed in https://github.com/espressif/esp-nn/commit/94399c921474fb6364e3985d6b82735eacb78c6a
Closing since there was no further confirmation.

closed 2025-09-02 by kedars