espressif/esp-nn #16: Arduino IDE support

eloquentarduino, 2024-09-13:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Provided a clear description of your suggestion.
- [X] Included any relevant context or examples.

### Issue or Suggestion Description

I'd like to add esp-nn optimizations in the Arduino IDE environment. I have a TFLite model exported as C header and I'm able to run it on my ESP32S3 "as-is", without esp-nn.
How do I make the Arduino IDE compile the esp-nn optimizations?

vikramdattu, 2025-03-28:
Hi, @eloquentarduino sorry for the delayed response.

`esp-nn` optimisations are by default part of the Arduino package with `esp-tflite-micro` component. Do you find it otherwise?

salernosimone, 2025-03-28:
I see (it's still me from a different account). You can close the issue

kedars, 2025-03-28:
Closing the issue