espressif/esp-nn #17: `esp-nn` 1.1.0 release on registry breaking older versions of `idf-component-manager`

kbx81, 2024-09-22:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Provided a clear description of your suggestion.
- [X] Included any relevant context or examples.

### Issue or Suggestion Description

We are using [esp-tflite-micro](https://github.com/espressif/esp-tflite-micro) which has a dependency on `esp-nn`; it appears that the recent release to `esp-nn` and corresponding update in the [registry](https://components.espressif.com/components/espressif/esp-nn/versions/1.1.0) is causing problems with older versions of `idf-component-manager`.

Specifically, `idf_component.yml` as pulled from the registry includes the `repository_info` key which is not recognized by the version of `idf-component-manager` we're using. It's noteworthy that this key is not present in the [`idf_component.yml` which appears in the repository](https://github.com/espressif/esp-nn/blob/v1.1.0/idf_component.yml); it only appears in the content pulled from the Espressif component registry.

Please [see our build logs here](https://github.com/esphome/esphome/actions/runs/10916163071/job/30336579740).

To confirm a workaround, I've forked [esp-tflite-micro](https://github.com/espressif/esp-tflite-micro) and tested two alternative dependency configurations:
1. I simply removed the `^` from the `version:` line within `dependencies`, forcing it to use the `1.0.0-rc1` which we know works.
2. I reconfigured the dependency to use the `esp-nn` GitHub repo directly (as opposed to the Espressif component registry) and explicitly specified `version: "v1.1.0"`.

Both of these alternate configurations resolved the error from `idf-component-manager`, of course at the expense of now having to use a fork (which we would prefer not to do).

It appears that moving to `platformio/espressif32` v6.6.0 with `framework-espidf` v5.2.1 or later gets us "compatible" `idf-component-manager` but we do not wish to make such a change at this time. Are there any other ways to fix this issue? Could we perhaps get a release of [framework-espidf](https://registry.platformio.org/tools/platformio/framework-espidf/versions) with IDF `v5.1.4` including a newer/compatible version of `idf-component-manager`?

vikramdattu, 2024-09-23:
Hi @kbx81 thanks for bringing this to light.

> Are there any other ways to fix this issue? Could we perhaps get a release of [framework-espidf](https://registry.platformio.org/tools/platformio/framework-espidf/versions) with IDF v5.1.4 including a newer/compatible version of `idf-component-manager`?

Newer releases do update the component manager version. Meaning, this will get updated with `v5.1.5`.

Easy way is to fix the component manager is to install the latest component manager using following pip command, after ESP-IDF setup is done:

`python -m pip install idf-component-manager`

kbx81, 2024-09-25:
Thank you for the reply @vikramdattu -- sorry for my delay circling back.

We are using platformio -- in short, ESPHome builds a platformio project with relevant source files and then hands it off to platformio to handle the build.

I have been experimenting a bit with using platformio's `extra_scripts` variable to call a "pre:" script. This works, but when `python -m pip install idf-component-manager` runs, it's installing into the environment in use _before_ the venv is created in which the build runs.

Take a look at the build logs [here](https://github.com/esphome/esphome/actions/runs/11010585866/job/30573095089?pr=7419#step:8:133) -- on line 132, we see the latest version of `idf-component-manager` is installed as expected by the "pre" script I added.

However, later, [on line 189](https://github.com/esphome/esphome/actions/runs/11010585866/job/30573095089?pr=7419#step:8:190), the build venv is created and the older version of `idf-component-manager` installed on [line 206](https://github.com/esphome/esphome/actions/runs/11010585866/job/30573095089?pr=7419#step:8:207). This older version is ultimately what gets used to attempt to retrieve `esp-nn`, which fails and derails the build.

Admittedly, platformio is a bit of a "black box" to me, so perhaps there is a way to fix this that I haven't learned about yet. A look through [platformio configuration](https://docs.platformio.org/en/latest/projectconf/sections/env/index.html) doesn't seem to reveal a way to hook into that part of the build. Do you have any suggestions as to how I might resolve this?

kbx81, 2024-09-25:
> Newer releases do update the component manager version. Meaning, this will get updated with `v5.1.5`.

It's looking more and more like this might be the easiest way for us to resolve this issue -- is there an ETA for a `v5.1.5` [framework-espidf](https://registry.platformio.org/tools/platformio/framework-espidf/versions) platformio package?

---

Totally unrelated -- what is the logic to what IDF versions are released on platformio? 😅 It seems like IDF v4.4.x patches are being released with greater frequency than IDF v5.x, which seems backwards at this point in time. 😇

vikramdattu, 2026-11-09:
Closing. Newer ESP-IDF releases ship a component manager that understands these manifest keys, so updating ESP-IDF (or `idf-component-manager`) resolves it.

closed 2026-11-09 by vikramdattu