espressif/esp-nn #19: Are the ESP-NN-optimizations available in ESP-DL?

nicklasb, 2025-03-09:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

Hi, 
Not sure if I am seeing any of the ESP32P4 (and S3) ESP-NN-optimizations used in ESP-DL? 
(or at least the optimization settings aren't visible as ESP-NN isn't used..)

vikramdattu, 2025-03-28:
Hi @nicklasb `esp-dl` is Espressif's own NN framework and hence, it has optimisations for its own kernels. The ESP-NN, however, is optimising the kernels for tflite kernels.

So, the options while using these will be different.

kedars, 2025-03-28:
Close #19

closed 2025-03-28 by kedars